facebook/openbmc #2: Won't compile:  Makefile:442: *** mixed implicit and normal rules.  Stop.

devicenull, 2015-03-17:
I'm running this on a CentOS 7 box (don't think that's relevant).  About halfway through bitbake, I'm getting the following:

<pre>
ERROR: Function failed: do_configure (log file is located at /home/brak/poky/build/tmp/work/wedge-poky-linux-gnueabi/linux-aspeed/2.6.28.9-r1/temp/log.do_configure.6767)
ERROR: Logfile of failure stored in: /home/brak/poky/build/tmp/work/wedge-poky-linux-gnueabi/linux-aspeed/2.6.28.9-r1/temp/log.do_configure.6767
Log data follows:
| DEBUG: Executing python function sysroot_cleansstate
| DEBUG: Python function sysroot_cleansstate finished
| DEBUG: Executing shell function do_configure
| NOTE: make wedge_defconfig
| Makefile:442: *** mixed implicit and normal rules.  Stop.
| ERROR: oe_runmake failed
| WARNING: /home/brak/poky/build/tmp/work/wedge-poky-linux-gnueabi/linux-aspeed/2.6.28.9-r1/temp/run.do_configure.6767:1 exit 1 from
|   exit 1
| ERROR: Function failed: do_configure (log file is located at /home/brak/poky/build/tmp/work/wedge-poky-linux-gnueabi/linux-aspeed/2.6.28.9-r1/temp/log.do_configure.6767)
ERROR: Task 71 (/home/brak/poky/meta-openbmc/meta-aspeed/recipes-kernel/linux/linux-aspeed_2.6.28.9.bb, do_configure) failed with exit code '1'
</pre>

tfangit, 2015-03-17:
Just based on "Makefile:442: *** mixed implicit and normal rules.  Stop.", I suspect that the make from CentOS 7 is new, and it does not like the Makefile from linux-2.6.28.9. 

In order to fix it, we will need to separate rules in Makefile:442. Meanwhile, if you can get an old make or CentOS 6.X, that would help to avoid this issue.

devicenull, 2015-03-17:
Same errors occurs on CentOS 6.6

devicenull, 2015-03-17:
These are the exact steps I did, starting with a clean CentOS install:

<pre>
yum install gawk make wget tar bzip2 gzip python unzip perl patch diffutils diffstat git cpp gcc gcc-c++ glibc-devel texinfo chrpath SDL-devel xterm
wget http://downloads.yoctoproject.org/releases/yocto/yocto-1.7.1/buildtools/poky-glibc-x86_64-buildtools-tarball-core2-64-buildtools-nativesdk-standalone-1.7.1.sh
bash poky-glibc-x86_64-buildtools-tarball-core2-64-buildtools-nativesdk-standalone-1.7.1.sh
adduser brak
su brak
cd ~
git clone -b daisy https://git.yoctoproject.org/git/poky
cd poky
git clone -b daisy https://github.com/openembedded/meta-openembedded.git
git clone https://github.com/facebook/openbmc.git meta-openbmc

cd ~/poky/; source /opt/poky/1.7.1/environment-setup-x86_64-pokysdk-linux; export TEMPLATECONF=meta-openbmc/meta-facebook/meta-wedge/conf; source oe-init-build-env
bitbake wedge-image</pre>

linlynn, 2015-03-26:
HI devicenull, I sent a pull request to fix this issue

tfangit, 2015-04-14:
@devicenull  Thx for the report. Please let us know if you still the issue after patch Lynn's diff.

@linlynn Thx for the patch.

linlynn, 2016-04-09:
the pr is merged so let us know if it still has issues

closed 2016-11-08 by theopolis

siutong, 2016-11-11:
I also found a build issue for compiling the images for Yosemite:
| Can't use 'defined(@array)' (Maybe you should just omit the defined()?) at /home/stong/openbmc/poky/build/tmp/work-shared/yosemite/kernel-source/kernel/timeconst.pl line 373.
| /home/stong/openbmc/poky/build/tmp/work-shared/yosemite/kernel-source/kernel/Makefile:122: recipe for target 'kernel/timeconst.h' failed
| make[3]: *** [kernel/timeconst.h] Error 255
| /home/stong/openbmc/poky/build/tmp/work-shared/yosemite/kernel-source/Makefile:861: recipe for target 'kernel' failed
| make[2]: *** [kernel] Error 2
| Makefile:128: recipe for target 'sub-make' failed
| make[1]: *** [sub-make] Error 2
| Makefile:19: recipe for target 'all' failed
| make: *** [all] Error 2
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/stong/openbmc/poky/build/tmp/work/yosemite-poky-linux-gnueabi/linux-aspeed/2.6.28.9-r2/temp/log.do_compile.22941)

This is due to perl 5.22 does not handle the statement, defined(@val):
Can't use 'defined(@array)' (Maybe you should just omit the defined()?) at timeconst.pl line 373.

This can be fixed by modifying defined(@val) as !@val.