facebook/openbmc #7: README and CONTRIBUTING docs say what you can contribute, but not how

rjek, 2015-03-31:
The "How can I contribute" section of README.md simply says what can be contributed, now how to contribute it.

While elsewhere it says you support git pulls, it would be nice to have information about any discussion mailing lists, IRC channels etc, where people can contribute things other than code as well as provide a forum for people getting into the code.

linlynn, 2015-04-02:
agree

rjek, 2015-04-14:
Having looked about for quite a bit, I can't find any public mailing list or IRC channel for the project: are there any that could be mentioned in the documentation?

tfangit, 2015-04-14:
@rjek We will set up something for the project. Will update README.md or CONTRIBUTING.md. Meanwhile, feel free to use issue system for any questions.