facebook/openbmc #8: New system overlay technique

causten, 2015-04-08:
I noticed you have wedge specific changes in the meta-aspeed directories... 
i.e. 0000-u-boot-aspeed-064.patch directs uboot to print to UART3... 
#define CONFIG_ASPEED_COM 0x1e784000 
i.e. 0000-linux-openbmc.patch disables the LPC

If I was to create a new system that uses UART1 and needs LPC from the Aspeed (i.e. slice) I was first thinking it should be meta-openbmc/meta-company/meta-slice, but that would pick up wedge specific changes for meta-aspeed.

What is your expectation for patch management and directory layout?

tfangit, 2015-04-14:
Ideally, board specific changes are in meta-openbmc/meta-company/meta-board. It is achievable but adding difficulty for development and maintenance, specially, in u-boot and kernel. Both u-boot and kernel already have great mechanism to support different boards/HWs. We would like leverage such mechanism, which means, HW/boards specific changes will be in SoC layer.

u-boot has board layers, currently, we re-used the ast2400 board layer for wedge development. That won't work for different boards. We will need to create separate directory for each different boards.

In kernel, we created separate configuration for wedge, wedge_defconfig. Then, use '#ifdef CONFIG_WEDGE' for wedge specific changes. However, we want to limit usage of '#ifdef' in source code. For example, for the code to enable/disable LPC, I would prefer to having a macro like CONFIG_ASPEED_ENABLE_LPC used in the source code, then each different HW defconfig has that macro defined or not depending on board specific design.

causten, 2015-04-15:
Sounds good, I'll get you some changes that adhere to the concepts.  Employing the _defconfig more fits with our thoughts too

closed 2015-04-15 by causten