facebook/openbmc #9: Licensing issues

Zarathecat, 2015-04-29:
Main issue: 

I'm interested in cloning and modifying this repo, but after checking 
the licensing, I'm worried that I can't legally do that yet. At the moment,
some makefiles and setup scripts say only 'Copyright Facebook.
All Rights Reserved.' (with no mention of the GPLv2), so I don't think 
I can currently distribute them.

If the intent of the licensing is to allow people to distribute and modify all 
files in the repo, then Facebook may wish to license those files differently
(eg: make it explicit within the files that they are licensed under GPLv2). 
I'm not an expert, but I'd rather not distribute/modify the
contents of the repo until this is clarified within the repo itself, and I 
think other potential contributors may feel similarly worried.

Other, smaller things: 

The'All Rights Reserved' statement no longer has legal weight and conflicts with
the GPL; see the conversation here:
https://lists.debian.org/debian-legal/2007/06/msg00253.html 
for more. The address of the Free Software Foundation is also out of date in
some files.

oridb, 2015-04-29:
Thanks for bringing this to our attention -- The intent is definitely to allow people to redistribute this code, and the license header is definitely something that we need to fix.

We're planning on a new push with several fixes in the near future -- I'll make sure we update the licenses to GPLv2 along with that.

Zarathecat, 2015-05-01:
Brilliant, thank you. :)

faybr, 2015-05-05:
Hi - do you have any plan for when the new push will be?

facebook-github-bot-3, 2015-08-04:
Thank you for reporting this issue and appreciate your patience. We've  notified the core team for an update on this issue. We're looking for a response within the next 30 days or the issue may be closed.

johnz00, 2015-10-01:
Hi, has there been any progress on updating the licenses, or with the latest push?  Thanks!

oridb, 2015-10-01:
If you look at the source files, you should see that the license is documented more clearly in the source files.

closed 2015-10-01 by oridb