facebook/openbmc #12: i2c-tools fetch error

fillmorechen, 2015-06-30:
Build Configuration:
BB_VERSION        = "1.22.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-12.04"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "wedge"
DISTRO            = "poky"
DISTRO_VERSION    = "1.6.3"
TUNE_FEATURES     = "armv5 thumb dsp arm926ejs"
TARGET_FPU        = "soft"
meta
meta-yocto
meta-yocto-bsp    = "daisy:240da756168b58eb05300a23c8282942f913db2d"
meta-oe
meta-networking   = "daisy:d3d14d3fcca7fcde362cf0b31411dc4eea6d20aa"
meta-openbmc
meta-aspeed
meta-wedge        = "master:2a51b7c1c2165ddb188c511e192b75f0aa0fbead"

NOTE: Preparing runqueue
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
WARNING: Failed to fetch URL svn://lm-sensors.org/svn/i2c-tools/branches/;protocol=http;module=i2c-tools-3.1, attempting MIRRORS if available
ERROR: Fetcher failure: Fetch command failed with exit code 1, output:
svn: E175002: Unable to connect to a repository at URL 'http://lm-sensors.org/svn/i2c-tools/branches/i2c-tools-3.1'
svn: E175002: OPTIONS of 'http://lm-sensors.org/svn/i2c-tools/branches/i2c-tools-3.1': could not connect to server (http://lm-sensors.org)

ERROR: Function failed: Fetcher failure for URL: 'svn://lm-sensors.org/svn/i2c-tools/branches/;protocol=http;module=i2c-tools-3.1'. Unable to fetch URL from any source.
ERROR: Logfile of failure stored in: /home/fillmore/poky/build/tmp/work/arm926ejste-poky-linux-gnueabi/i2c-tools/3.1.1-r0/temp/log.do_fetch.9458
ERROR: Task 326 (/home/fillmore/poky/meta-openbmc/common/recipes-core/i2c-tools/i2c-tools_3.1.1.bb, do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2548 tasks of which 2547 didn't need to be rerun and 1 failed.
No currently running tasks (1701 of 2561)

Summary: 1 task failed:
  /home/fillmore/poky/meta-openbmc/common/recipes-core/i2c-tools/i2c-tools_3.1.1.bb, do_fetch
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
fillmore@SDD2:~/poky/build$

fillmorechen, 2015-07-08:
Issue fixed, Thanks

tfangit, 2015-07-23:
Thx. Most likely, there was an issue to access the i2c-tool source, which was fixed by itself later. 

For 3rd party open source projects, OpenBMC repo (Yocto) only has the reference to its source. The source code will be downloaded during build.

closed 2015-07-23 by tfangit

wvervoorn, 2015-10-15:
I am experiencing the same issue. It seems lm-server.org is down and so far I haven't found an alternative i2ctools repo that works as well. I found the https://github.com/groeck/i2c-tools.git repo but I still get a build error in the impbd stuff as there are no defines for the I2C slave in the i2c-dev.h.

Any suggestions

williamspatrick, 2015-10-15:
Yocto keeps mirrors of packages here : http://downloads.yoctoproject.org/mirror/sources/

The i2c-tools-3.1.1.tar.bz2 is there.

wvervoorn, 2015-10-16:
Hello Patrick,

Thanks for that. Will the OpenBMC tree be updated to include these mirrors at least for the i2c-tools case?

In the meantime I got it working by correctly getting the  code from another GIT repo on github.

I also mentioned an issue building ipmbd that I expected to be related to this issue. It wasn’t. Actually this was caused by the order in which the bitbake tasks are executed. The ipmbd code requires “facebook/i2c-dev.h” to be present and this was installed much later in the process. Perhaps this was a coincidence on my build. It would be better to know for sure that one task can only be executed when another has been completed.

Best Regards,
Wim Vervoorn

Eltan B.V.
Ambachtstraat 23
5481 SM Schijndel
The Netherlands

T : +31-(0)73-594 46 64
E : wvervoorn@eltan.com
W : http://www.eltan.com
"THIS MESSAGE CONTAINS CONFIDENTIAL INFORMATION. UNLESS YOU ARE THE INTENDED RECIPIENT OF THIS MESSAGE, ANY USE OF THIS MESSAGE IS STRICTLY PROHIBITED. IF YOU HAVE RECEIVED THIS MESSAGE IN ERROR, PLEASE IMMEDIATELY NOTIFY THE SENDER BY TELEPHONE +31-(0)73-5944664 OR REPLY EMAIL, AND IMMEDIATELY DELETE THIS MESSAGE AND ALL COPIES."

williamspatrick, 2015-10-16:
On Fri, Oct 16, 2015 at 12:03:29AM -0700, Wim Vervoorn wrote:
> Hello Patrick,
> 
> Thanks for that. Will the OpenBMC tree be updated to include these mirrors at least for the i2c-tools case?

I cannot answer this.  I am not working on this tree.

> I also mentioned an issue building ipmbd that I expected to be related to this issue. It wasn’t. Actually this was caused by the order in which the bitbake tasks are executed. The ipmbd code [...]

It appears that i2c-dev.h is provided by the package 'fbutils'.  I don't
see that in the dependency list for the ipmbd package. My suggestion would
be to add DEPENDS += "fbutils" to common/recipes-core/ipmbd/ipmbd_0.1.bb

You could try this out and make a pull request for Facebook to
incorporate.
-- 
Patrick Williams

wvervoorn, 2015-10-16:
Hello Patrick,

Thanks for the quick reply. I will do that.

Best Regards,
Wim Vervoorn

Eltan B.V.
Ambachtstraat 23
5481 SM Schijndel
The Netherlands

T : +31-(0)73-594 46 64
E : wvervoorn@eltan.com
W : http://www.eltan.com
"THIS MESSAGE CONTAINS CONFIDENTIAL INFORMATION. UNLESS YOU ARE THE INTENDED RECIPIENT OF THIS MESSAGE, ANY USE OF THIS MESSAGE IS STRICTLY PROHIBITED. IF YOU HAVE RECEIVED THIS MESSAGE IN ERROR, PLEASE IMMEDIATELY NOTIFY THE SENDER BY TELEPHONE +31-(0)73-5944664 OR REPLY EMAIL, AND IMMEDIATELY DELETE THIS MESSAGE AND ALL COPIES."