facebook/openbmc #15: Clone meta-openbmc error: Permission Denied (publickey)

hoanganhbui, 2015-08-31:
I can't clone repository meta-openbmc with
git clone git@github.com:facebook/openbmc.git meta-openbmc

It say that:
anhbh@ubuntu:~/wedge/poky$ git clone git@github.com:facebook/openbmc.git meta-openbmc
Cloning into 'meta-openbmc'...
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

Anyone has solution? Tks.

shenki, 2015-08-31:
There are two solutions:

1. Add your ssh public key to your github account
2. Clone from https instead: ```git clone https://github.com/facebook/openbmc meta-openbmc```

hoanganhbui, 2015-08-31:
Solved! thank you Shenki

tfangit, 2015-08-31:
Thx, @shenki .

closed 2015-08-31 by tfangit