facebook/openbmc #19: Permission denied

liufujun135, 2015-12-07:
Hi All,

I can not clone the openbmc codes. The issue is "Permission denied ". 

XXX@DZ-PC-099:~/openbmc$ git clone git@github.com:facebook/openbmc.git
Cloning into 'openbmc'...
Warning: Permanently added the RSA host key for IP address '192.30.252.130' to the list of known hosts.
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

oridb, 2015-12-07:
You seem to have incorrectly configured SSH keys for github. You can clone over HTTP, or follow the guide for setting up your github account for SSH access here:

https://help.github.com/articles/generating-ssh-keys/

williamspatrick, 2016-01-09:
Please close issue.

closed 2016-01-09 by oridb