facebook/openbmc #21: Install instructions do not work.

usernameisalreadytaken2014, 2016-03-04:
As per instructions in README.md:
> $ git clone git@github.com:facebook/openbmc.git meta-openbmc

Expected:
A git clone...

Actual:

> Cloning into 'meta-openbmc'...
> Warning: Permanently added the RSA host key for IP address '192.30.252.131' to the list of known hosts.
> Permission denied (publickey).
> fatal: Could not read from remote repository.

theopolis, 2016-03-04:
Good catch! This needs to be updated to say: 
```
$ git clone https://github.com/facebook/openbmc
```

mkumatag, 2016-03-04:
It does work but you should have ssh key added for your machine:

[manjunath@oc1275007250 fb]$ git clone git@github.com:facebook/openbmc.git meta-openbmc
Initialized empty Git repository in /home/manjunath/fb/meta-openbmc/.git/
remote: Counting objects: 1167, done.
remote: Total 1167 (delta 0), reused 0 (delta 0), pack-reused 1167
Receiving objects: 100% (1167/1167), 1.47 MiB | 475 KiB/s, done.
Resolving deltas: 100% (346/346), done.
[manjunath@oc1275007250 fb]$ ll
total 4
drwxr-xr-x. 7 manjunath manjunath 4096 Mar  4 09:11 meta-openbmc
[manjunath@oc1275007250 fb]$

theopolis, 2016-04-06:
Hey @usernameisalreadytaken2014, thanks for your patience, the README should be clearer now, please let us know if there are still confounds/errors. :dancer:

closed 2016-04-06 by theopolis