facebook/openbmc #22: Clone "meta-openbmc" failed

jy03163035, 2016-03-11:
Dear sirs,

I tried to get the code with the method in README.md, but failed. Seems that I have no rights to access the server.  Could you help me about this issue? Thank you very much.

git clone git@github.com:facebook/openbmc.git meta-openbmc
Cloning into 'meta-openbmc'...
Permission denied (publickey).
fatal: The remote end hung up unexpectedly

BR,
Andy

theopolis, 2016-03-11:
Hi @jy03163035, thanks for the note, check out the comment here: https://github.com/facebook/openbmc/issues/21#issuecomment-192070763

The instructions in the README will be updated soon!

jy03163035, 2016-03-16:
Thanks for your replay. 
So if I haven't finished Step 0, I will get failure in Step 2, right?

theopolis, 2016-04-06:
@Andywjzhang, a few of the recent updates to the README should clear things up, checkout the new content from: https://github.com/facebook/openbmc/commit/0f6e01e5b8f508daf3867beb59a2631a3fcab77f. :+1:

closed 2016-04-06 by theopolis