facebook/openbmc #30: over lan

yayake, 2016-04-12:
the openbmc seems only support IPMI's KCS messge interface, about other message interface,such as LAN, serial/modem. While openbmc may support SOL, without lan and serial interface, the SOL function seem useless.

williamspatrick, 2016-04-12:
We have implemented an application that can route the VUART of the host
to a number of different locations.  See
https://github.com/openbmc/obmc-console .

I believe we currently use this to route over an SSH connection, but
@jk-ozlabs could fill in more details on what we currently do and what
we are considering doing in the future.

yayake, 2016-04-19:
e, over an ssh connection?
why not implements RMCP+ protocol, according to standard IPMI  v2.0.
recently, I'm little confuse about the function of openbmc's project, compared to bmc running in server.
Openbmc is only focused on little witch box?

williamspatrick, 2016-04-28:
The current implementations did not have a need (or desire) for IPMI, due to the data centers they were being deployed into.  We recognize for a general-purpose / commercial server market IPMI is desirable but it has not been implemented yet.

theopolis, 2016-11-08:
I'm closing this issue since the discussion has trailed off.

closed 2016-11-08 by theopolis