facebook/openbmc #31: bitbake failure due to explicit protocol git in u-boot recipes not supported by firewall

an0maly, 2016-04-14:
Not all corporate firewalls support git protocol, nor do gitproxy configuration/scripts necessarily work.  However, http proxy usually is available.   

Modifying the meta-aspeed/recipes-bsp/u-boot/u-boot-fw-utils_2013.07.bb and meta-aspeed/recipes-bsp/u-boot/u-boot_2013.07.bb recipes with protocol=http resolved this issue.

theopolis, 2016-04-15:
Thanks @an0maly, will fix!

bcg62, 2016-10-31:
I'll open a separate issue but hit the same problem here

ERROR: Function failed: Fetcher failure for URL: 'git://github.com/theopolis/linux.git;branch=openbmc/fido/2.6.28;protocol=git'. Unable to fetch URL from any source.

theopolis, 2016-11-08:
This should be fixed with 87e27c1aff7b266d7adc7370c069de274dee1e55, feel free to reopen or ping me if there are still issues. :)

closed 2016-11-08 by theopolis

bcg62, 2016-11-08:
@theopolis got another one here.

WARNING: Failed to fetch URL git://sourceware.org/git/glibc.git;branch=release/2.21/master;name=glibc, attempting MIRRORS if available

theopolis, 2016-11-09:
Ok, that's outside the scope of our `meta-` layer. Did `bitbake` succeed using the mirrors `glibc` defined?

bcg62, 2016-11-09:
You're right it did, sorry spoke too soon.