facebook/openbmc #32: How to flash the OpenBMC build image into an actual BMC on an off-the-shelf motherboard?

phamtson, 2016-06-21:
Hi,

What is the tool to flash the OpenBMC build image into a BMC of an off-the-shelf motherboard?

tfangit, 2016-06-21:
OpenBMC image is associated with the board which the image is build for. With some minimum changes (i.e. change console UART port), you can get the OpenBMC image that could boot on that board. At that time, you can flash the whole image to the SPI flash (if you can flash it through an external flasher) and start from there.

closed 2016-06-21 by tfangit

lgon, 2017-06-07:
A way to free the memory without restarting phosphor-gevent.service is to truncate the file marked for deletion. 555 is the pid of "/usr/sbin/phosphor-gevent rest_dbus"
```root@qemuarm:/proc/555/fd# df
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root                59365     44448     10331  81% /
devtmpfs                124196         0    124196   0% /dev
tmpfs                   124404         0    124404   0% /dev/shm
tmpfs                   124404     24928     99476  20% /run
tmpfs                   124404         0    124404   0% /sys/fs/cgroup
tmpfs                   124404      8032    116372   6% /tmp
tmpfs                   124404       228    124176   0% /var/volatile
root@qemuarm:/proc/555/fd# ls -la /proc/555/fd 
total 0
dr-x------    2 root     root             0 Jun  7 20:35 .
dr-xr-xr-x    9 root     root             0 Jun  7 20:35 ..
lr-x------    1 root     root            64 Jun  7 20:35 0 -> /dev/null
lrwx------    1 root     root            64 Jun  7 20:35 1 -> socket:[28071]
lrwx------    1 root     root            64 Jun  7 20:35 11 -> /tmp/tmp4GcGr4 (deleted)
lrwx------    1 root     root            64 Jun  7 20:35 12 -> /tmp/tmpVe2PVG (deleted)
lrwx------    1 root     root            64 Jun  7 20:35 2 -> socket:[28071]
lrwx------    1 root     root            64 Jun  7 20:35 3 -> socket:[9668]
lrwx------    1 root     root            64 Jun  7 20:35 4 -> socket:[28270]
lrwx------    1 root     root            64 Jun  7 20:35 5 -> socket:[9668]
lrwx------    1 root     root            64 Jun  7 20:35 6 -> anon_inode:[eventpoll]
lrwx------    1 root     root            64 Jun  7 20:35 7 -> anon_inode:[eventfd]
lr-x------    1 root     root            64 Jun  7 20:35 8 -> /dev/urandom
root@qemuarm:/proc/555/fd# : > /proc/555/fd/12
root@qemuarm:/proc/555/fd# df
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root                59365     44448     10331  81% /
devtmpfs                124196         0    124196   0% /dev
tmpfs                   124404         0    124404   0% /dev/shm
tmpfs                   124404     24928     99476  20% /run
tmpfs                   124404         0    124404   0% /sys/fs/cgroup
tmpfs                   124404         0    124404   0% /tmp
tmpfs                   124404       292    124112   0% /var/volatile
```