facebook/openbmc #36: Differences between facebook/openbmc and openbmc/openbmc

ruchi393, 2016-09-20:
For openBMC, there seem to be two yocto based projects -
facebook/openbmc
openbmc/openbmc

How similar or how different are these ? Are there any plans to merge the two efforts ? Can the utilities from openbmc/openbmc be used with this openbmc project and vica versa ?

tfangit, 2016-09-21:
You can think these two are two different OpenBMC distributions. 

IBM team is the main driving force behind openbmc/openbmc. We are cooperating with IBM team on u-boot and Linux kernel areas. On user application side, each team implemented various different utilities for different HWs. If you get all dependencies pulled in, a utility from one project can be used in another project. However, some utilities that have HW dependency will have some difficulty to port to another HW. That's one of the areas we are trying to make it better.

closed 2016-09-21 by tfangit

pabbachit, 2016-09-21:
Tian,

We are trying to get and build Opnebmc/Openbmc version stack for
AST2400/2500 but unfortunately we are not able to find any info to build
OpenBMC stack for AST part. Can you please guide to build Open BMC stack
for AST2400 SOC?

theopolis, 2016-09-22:
Hi @pabbachit we can definitely help with building `facebook/openbmc` for the AST2400. I would open an issue on [`openbmc/openbmc`](https://github.com/openbmc/openbmc) if you're trying to build that repo.

williamspatrick, 2016-09-22:
@pabbachit - most of the machines supported by openbmc/openbmc are AST2400 based. We have instructions in the README for building them. Building for a raw AST2400 dev board should be as simple as setting the TEMPLATECONF environment variable to the right value. As @theopolis mentioned you can open an issue there if you have trouble or else we have members in most time zones on Freenet IRC #openbmc where you can get active support.

manojkiraneda, 2017-03-17:
@williamspatrick - I hav set teh TEMPLATECONF to plametto for kernel build.Is there any similar command like menuconfig through which we can get the custom image by removing some of the peripherals that are not required.

williamspatrick, 2017-03-20:
@manojkiraneda - We should probably take any openbmc/openbmc discussion to that project and not here.  Since it is relevant for both facebook/openbmc and openbmc/openbmc, as they both use Yocto as their underlying distro, I will answer this question here.  

I am not aware of any menuconfig-like functionality for Yocto as a whole, but the kernel portion does have that functionality.  You might reference this document from Yocto: http://www.yoctoproject.org/docs/2.2/kernel-dev/kernel-dev.html

Something like 'bitbake linux-obmc -c menuconfig' may be what you are looking for.  Generally we use config snippets to make modifications to the base kconfig and place those into the bitbake tree.  Discussion of config snippets is also found in that kernel-dev document.

jerrchen, 2017-09-25:
Will ever openbmc/openbmc and facebook/openbmc be converged to one distribution? or will it forever be operating in parallel?