facebook/openbmc #37: perl-5.20.0-r1 package failing

bcg62, 2016-10-31:
Slightly unfamiliar with yocto but can't seem to get past this in the build process.

I'm building inside a centos 6.3 vagrant project which i can share if that helps.

```
ERROR: objcopy failed with exit code 256 (cmd was 'arm-poky-linux-gnueabi-objcopy' --only-keep-debug '/mnt/build/openbmc/poky/build/tmp/work/armv5e-poky-linux-gnueabi/perl/5.20.0-r1/package/usr/lib/perl/ptest/generate_uudmap' '/mnt/build/openbmc/poky/build/tmp/work/armv5e-poky-linux-gnueabi/perl/5.20.0-r1/package/usr/lib/perl/ptest/.debug/generate_uudmap'):
arm-poky-linux-gnueabi-objcopy:/mnt/build/openbmc/poky/build/tmp/work/armv5e-poky-linux-gnueabi/perl/5.20.0-r1/package/usr/lib/perl/ptest/generate_uudmap: File format not recognized
Function failed: split_and_strip_files
```

```
DEBUG: Executing python function sstate_task_prefunc
DEBUG: Python function sstate_task_prefunc finished
DEBUG: Executing python function do_package
DEBUG: Executing python function package_get_auto_pr
DEBUG: Python function package_get_auto_pr finished
DEBUG: Executing python function perform_packagecopy
DEBUG: Python function perform_packagecopy finished
DEBUG: Executing shell function perl_package_preprocess
DEBUG: Shell function perl_package_preprocess finished
DEBUG: Executing python function split_and_strip_files
ERROR: objcopy failed with exit code 256 (cmd was 'arm-poky-linux-gnueabi-objcopy' --only-keep-debug '/mnt/build/openbmc/poky/build/tmp/work/armv5e-poky-linux-gnueabi/perl/5.20.0-r1/package/usr/lib/perl/ptest/generate_uudmap' '/mnt/build/openbmc/poky/build/tmp/work/armv5e-poky-linux-gnueabi/perl/5.20.0-r1/package/usr/lib/perl/ptest/.debug/generate_uudmap'):
arm-poky-linux-gnueabi-objcopy:/mnt/build/openbmc/poky/build/tmp/work/armv5e-poky-linux-gnueabi/perl/5.20.0-r1/package/usr/lib/perl/ptest/generate_uudmap: File format not recognized
DEBUG: Python function split_and_strip_files finished
DEBUG: Python function do_package finished
ERROR: Function failed: split_and_strip_files
```

theopolis, 2016-11-08:
Interesting bug, unfortunately I've never encountered. While there aren't many reporting this bug with perl, there are some. ;) I wonder if it's related to your build environment and `gcc6`?

I usually build on Ubuntu 14.04 and 16.04, CentOS6.3 is old and shouldn't have a 'too-new' of a compiler. Can you try on a CentOS 7?

bcg62, 2016-11-08:
Thanks @theopolis - testing 7.2 now.

I originally started with 7.2 but encountered this warning which sent me down the 6.3 rabbit hole

```
WARNING: Host distribution "CentOS-Linux-7.2.1511" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
```

Maybe this should be updated?

theopolis, 2016-11-08:
Right, ;) I just ignore it, haven't investigated where it comes from.

bcg62, 2016-11-09:
The issue isn't present with 7.2, completed a full build successfully.

theopolis, 2016-11-11:
Woot! We'll rebase and start using a newer version of Yocto soon, which should remove the CentOS7 warning. But that's what I use to build normally-- consider the warning nothing more than an annoyance.

closed 2016-11-11 by theopolis