facebook/openbmc #38: Arrow keys not working properly in mTerm_client

wililupy, 2016-12-19:
Hello,

I am running mTerm_client on my OpenBMC and everytime I try to use the arrow keys, either for recalling previous command or to make selections in the BIOS, it drops me out of mTerm_client much like pressing CTRL-X and I am back in the root@bmc shell. 

Not seeing where in the code this is breaking. When I try to connect with microcom, the display is unusable, but the arrow keys appear to be actually working instead of dropping me back to the bmc shell.

tfangit, 2016-12-19:
The latest mTerm fixed such issue and also added new feature like being able to send 'break'. Looks like the OpenBMC release shipped with your HW is an old release. Want to try to build based on the latest and flash it?

wililupy, 2016-12-19:
I will contact the HW Vendor to see if they have a newer version. Lightning is currently on the device. I worked around it by killing mTerm_server and then re-running sol.sh and I could use the arrow keys again.
I may try to build and install a newer version sometime down the road. We can close this issue and I'll create a new one if I have to.

Thanks
Luke

closed 2016-12-19 by wililupy