facebook/openbmc #43: ftgmac error

karthikbgt, 2017-04-11:
Hi,

I get the attached error when i try to do an SCP of a file via ethernet interface to OpenBMC running in AST2520 from an external server. So far, i have faced this issue whenever i try to do a SCP.
[ErrorLog.txt](https://github.com/facebook/openbmc/files/914444/ErrorLog.txt)

Can you kindly confirm which version of OpenBMC addresses this ftgmac issue.

Kindly note that I get the error in system that is Linux bmc 4.1.15 armv6l GNU/Linux . The file /etc/os-release does not seem to exist in the system. However, the /etc/issue seems to convey as 'OpenBMC Release cmm-v11' ... 

Can you kindly provide me some pointers on what should be taken care and steps to overcome this issue. 

Thanks in advance,
Karthik Balaguru

karthikbgt, 2017-04-18:
Also, it is observed that whenever the ftgmac rx buf exhaustion & packet loss occurs in the CMM's(OpenBMC 4.1.15) management interface, the interface seems to reach a weird state/go for a toss and it does not recover if i try ifdown/ifup or ifconfig eth0 down/up or even for soft reboot. It recovers only if i powercycle the CMM. 

Is there any other way to overcome this packet loss and also ensure that the interface does not go down ?

tfangit, 2017-04-19:
@karthikbgt Thx for reporting the issue. We do notice such error also during scp. However, we don't see any other side effect other than the error message. How big is the file you are copying?

SCM does not go through the CMM MAC to communicate to outside. There are an internal 16 HW switch connecting SCM to the mgmt port. CMM MAC is just another port on that switch. So, this error message, which comes from the CMM MAC driver, shall not impact SCM networking. Could u provide more details on your concern about SCM networking in this case?

karthikbgt, 2017-04-19:
@tfangit Thanks for confirming that the such error are observed while SCP. The size of file that i tried was ~ 409M.  Glad to know that this error message will not have any side effect and also it will not impact SCM networking.

closed 2019-08-03 by jda