facebook/openbmc #47: u-boot build issue

AdrianB-Flex, 2017-05-02:
Seeing this problem while building wedge100 with no local changes:

[image]

AdrianB-Flex, 2017-05-02:
Building on a Ubuntu 14.04.5 machine and following the howto build section of this repo.

theopolis, 2017-05-04:
Hi @AdrianB-Flex, we can help, but I need more details about the error :)

theopolis, 2017-05-04:
Here's the build that continually tries to build `wedge100-image` from the `helium` branch: https://jenkins.osquery.io/view/Other/job/openbmcHeliumBuildWedge100/

It runs:
```
export BOARD=wedge100
export OPENBMC=$PWD

sudo yum install -y perl-Thread-Queue gcc-c++ patch wget diffstat chrpath texinfo bzip2

git clone -b krogoth --depth 1 https://git.yoctoproject.org/git/poky
cd poky
git clone -b krogoth --depth 1 https://github.com/openembedded/meta-openembedded.git
git clone -b krogoth --depth 1 https://git.yoctoproject.org/git/meta-security
ln -s $OPENBMC meta-openbmc

export TEMPLATECONF=meta-openbmc/meta-facebook/meta-${BOARD}/conf
source ./oe-init-build-env

bitbake ${BOARD}-image
```

AdrianB-Flex, 2017-05-08:
Sorry for the delay Teddy, I was on PTO latter half of last week.  I am using fido not krogoth, that might be an issue.  I'll give this a try.

AdrianB-Flex, 2017-05-08:
Not seeing the meta-openbmc/meta-facebook folder.  Is there a git clone missing?  Before I had to do this:
git clone https://github.com/facebook/openbmc.git meta-openbmc

theopolis, 2017-05-10:
Here's the `meta-facebook` folder: https://github.com/facebook/openbmc/tree/helium/meta-facebook

Would you mind starting from a clean working state and documenting the steps you're taking, I think that will help me debug too :)

amithash, 2017-05-10:
@AdrianB-Flex if you are moving to krogoth from fido, you might also want to switch openbmc from the master to the helium branch. 
git clone -b helium https://github.com/facebook/openbmc.git meta-openbmc
instead of
git clone https://github.com/facebook/openbmc.git meta-openbmc

AdrianB-Flex, 2017-05-10:
Ahh, thanks Amit.

AdrianB-Flex, 2017-05-11:
From a clean tree Ubuntu 14.04.5:
sudo apt-get install gawk wget git-core diffstat unzip texinfo gcc-multilib \
     build-essential chrpath libsdl1.2-dev xterm

git clone -b krogoth --depth 1 https://git.yoctoproject.org/git/poky bolt-dev
cd bolt-dev
git clone -b krogoth --depth 1 https://github.com/openembedded/meta-openembedded.git
git clone -b krogoth --depth 1 https://git.yoctoproject.org/git/meta-security
git clone -b helium https://github.com/facebook/openbmc.git meta-openbmc
export BOARD=wedge100
export TEMPLATECONF=meta-openbmc/meta-facebook/meta-${BOARD}/conf
source ./oe-init-build-env
bitbake ${BOARD}-image

AdrianB-Flex, 2017-05-11:
And here's the error:

[image]

tfangit, 2017-05-11:
@AdrianB-Flex The default gcc from poky krogoth branch is 5.x. However, your log shows as 4.9.2. Could u pls check why?

AdrianB-Flex, 2017-05-11:
That's puzzling.  When I do a gcc --version I get 6.2.0.  This got pulled in from the apt-get install gcc-multilib from the yocto user guide.  Not sure why 4.9.2 is getting selected.

theopolis, 2017-05-12:
I'll try to reproduce on a Ubuntu 14.04 system tonight.

FrankHome, 2017-05-12:
When I move to u-boot path and
'make fbwedge100_defconfig' && 'make'
also get this same problem:
[image]
How to fix the Assembler Error?

theopolis, 2017-05-12:
Right, so this is a compiler issue. Is there the older GCC in your path?

On a new install of Ubuntu 14.04: https://jenkins.osquery.io/view/Other/job/openbmcUbuntuHeliumBuildWedge100/8/console

Using the build script from above.

FrankHome, 2017-05-13:
Thanks a lot , at first I use the poky fido branch,the gcc version is 4.9 .But now I use the krogoth branch, all going well,it can build success.

AdrianB-Flex, 2017-05-15:
Frank, what distro/version did you use finally?

FrankHome, 2017-05-15:
Ubuntu-14.04.1-desktop-i386

Be honest, I did not find the real reason why I  can build success finally, but , These are the steps I use to get things done:
sudo apt-get insatll gcc-arm-linux-gnueabi
sudo apt-get insatll g++-arm-linux-gnueabi
sudo apt-get install sed wget subversion git-core coreutils unzip gcc-multilib \ 
texi2html texinfo libsdl1.2-dev docbook-utils fop gawk python-pysqlite2 \
diffstat make gcc build-essential xsltproc g++ desktop-file-utils \
chrpath libgl1-mesa-dev libglu1-mesa-dev autoconf automake groff libtool \
xterm libxml-parser-perl vim-common xz-utils cvs tofrodos \
libstring-crc32-perl ubuntu-minimal ubuntu-standard patch \
libbonobo2-common libncurses5-dev \

git clone -b krogoth --depth 1 https://git.yoctoproject.org/git/poky
cd poky
git clone -b krogoth --depth 1 https://github.com/openembedded/meta-openembedded.git
git clone -b krogoth --depth 1 https://git.yoctoproject.org/git/meta-security
git clone -b helium https://github.com/facebook/openbmc.git meta-openbmc

source  oe-init-build-env
bitbake -k wedge100-image

tfangit, 2017-05-16:
Switching to krogoth and start a new build directory shall fix this issue, as gcc 5.x is used in Krogoth. 

@AdrianB-Flex  Could you confirm if this fixed your build issue?

AdrianB-Flex, 2017-05-17:
Confirmed.  Everyone, thanks for the help.  The "How to build" section on the main facebook/openbmc page should be updated I think.

closed 2017-05-18 by AdrianB-Flex