facebook/openbmc #48: Build U-boot faild

FrankHome, 2017-05-10:
#47 
When run "bitbake wedge100-image " or "bitbake u-boot" the issue will appear
please help me !!!

[image]
[image]

theopolis, 2017-05-10:
Hello! There's not enough information included in your screen shot to help me help you debug. Can you please include more details about what specifically is causing the compile errors?

FrankHome, 2017-05-10:
HI，Theopolis

Bellow is the all log when i run "bitbake u-boot" 
please help analyse ,Thanks a lot :

parallels@ubuntu:~/openBmc/poky/build$ bitbake u-boot
WARNING: Host distribution "Ubuntu-16.04" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |###########################################| ETA:  00:00:00
Loaded 2315 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION        = "1.26.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-16.04"
TARGET_SYS        = "arm-fb-linux-gnueabi"
MACHINE           = "wedge"
DISTRO            = "poky"
DISTRO_VERSION    = "0.4"
TUNE_FEATURES     = "arm armv5 thumb dsp"
TARGET_FPU        = "soft"
meta              
meta-yocto        
meta-yocto-bsp    = "fido:c53ddb2006f04051b00df3189fe8d35bb49ef3c7"
meta-oe           
meta-networking   
meta-python       
meta-perl         = "fido:902964a4da26e46018d2a8d17dcdda1ac4627a39"
meta-openbmc      = "helium:beaa287b118eb5de7d764eb62c6adb70f03075b2"
meta-security     = "fido:aac972c0b88ce67a7bc9fcd83ee8793b61d4fff7"
meta-aspeed       
meta-facebook     
meta-wedge        
meta-wedge100     = "helium:beaa287b118eb5de7d764eb62c6adb70f03075b2"

NOTE: Preparing RunQueue
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: Function failed: do_compile (log file is located at /home/parallels/openBmc/poky/build/tmp/work/armv5e-fb-linux-gnueabi/u-boot/v2016.07-r0/temp/log.do_compile.10395)
ERROR: Logfile of failure stored in: /home/parallels/openBmc/poky/build/tmp/work/armv5e-fb-linux-gnueabi/u-boot/v2016.07-r0/temp/log.do_compile.10395
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 2 CROSS_COMPILE=arm-fb-linux-gnueabi- CC=arm-fb-linux-gnueabi-gcc  --sysroot=/home/parallels/openBmc/poky/build/tmp/sysroots/wedge V=1 HOSTCC=gcc  -isystem/home/parallels/openBmc/poky/build/tmp/sysroots/x86_64-linux/usr/include -O2 -pipe -L/home/parallels/openBmc/poky/build/tmp/sysroots/x86_64-linux/usr/lib -L/home/parallels/openBmc/poky/build/tmp/sysroots/x86_64-linux/lib -Wl,-rpath-link,/home/parallels/openBmc/poky/build/tmp/sysroots/x86_64-linux/usr/lib -Wl,-rpath-link,/home/parallels/openBmc/poky/build/tmp/sysroots/x86_64-linux/lib -Wl,-rpath,/home/parallels/openBmc/poky/build/tmp/sysroots/x86_64-linux/usr/lib -Wl,-rpath,/home/parallels/openBmc/poky/build/tmp/sysroots/x86_64-linux/lib -Wl,-O1 O=default fbwedge_config
| make -C /home/parallels/openBmc/poky/build/tmp/work/armv5e-fb-linux-gnueabi/u-boot/v2016.07-r0/git/default KBUILD_SRC=/home/parallels/openBmc/poky/build/tmp/work/armv5e-fb-linux-gnueabi/u-boot/v2016.07-r0/git \
| -f /home/parallels/openBmc/poky/build/tmp/work/armv5e-fb-linux-gnueabi/u-boot/v2016.07-r0/git/Makefile fbwedge_config
| make[1]: Entering directory '/home/parallels/openBmc/poky/build/tmp/work/armv5e-fb-linux-gnueabi/u-boot/v2016.07-r0/git/default'
| make -f ../scripts/Makefile.build obj=scripts/basic
| ln -fsn .. source
| rm -f .tmp_quiet_recordmcount
| /bin/bash ../scripts/mkmakefile \
|     .. . 2016 07
|   GEN     ./Makefile
| make -f ../scripts/Makefile.build obj=scripts/kconfig fbwedge_config
| scripts/kconfig/conf  --defconfig=arch/../configs/fbwedge_defconfig Kconfig
| #
| # configuration written to .config
| #
| make[1]: Leaving directory '/home/parallels/openBmc/poky/build/tmp/work/armv5e-fb-linux-gnueabi/u-boot/v2016.07-r0/git/default'
| NOTE: make -j 2 CROSS_COMPILE=arm-fb-linux-gnueabi- CC=arm-fb-linux-gnueabi-gcc  --sysroot=/home/parallels/openBmc/poky/build/tmp/sysroots/wedge V=1 HOSTCC=gcc  -isystem/home/parallels/openBmc/poky/build/tmp/sysroots/x86_64-linux/usr/include -O2 -pipe -L/home/parallels/openBmc/poky/build/tmp/sysroots/x86_64-linux/usr/lib -L/home/parallels/openBmc/poky/build/tmp/sysroots/x86_64-linux/lib -Wl,-rpath-link,/home/parallels/openBmc/poky/build/tmp/sysroots/x86_64-linux/usr/lib -Wl,-rpath-link,/home/parallels/openBmc/poky/build/tmp/sysroots/x86_64-linux/lib -Wl,-rpath,/home/parallels/openBmc/poky/build/tmp/sysroots/x86_64-linux/usr/lib -Wl,-rpath,/home/parallels/openBmc/poky/build/tmp/sysroots/x86_64-linux/lib -Wl,-O1 O=default all
| make -C /home/parallels/openBmc/poky/build/tmp/work/armv5e-fb-linux-gnueabi/u-boot/v2016.07-r0/git/default KBUILD_SRC=/home/parallels/openBmc/poky/build/tmp/work/armv5e-fb-linux-gnueabi/u-boot/v2016.07-r0/git \
| -f /home/parallels/openBmc/poky/build/tmp/work/armv5e-fb-linux-gnueabi/u-boot/v2016.07-r0/git/Makefile all
| make[1]: Entering directory '/home/parallels/openBmc/poky/build/tmp/work/armv5e-fb-linux-gnueabi/u-boot/v2016.07-r0/git/default'
| make -f ../Makefile silentoldconfig
| make -f ../scripts/Makefile.build obj=scripts/basic
| ln -fsn .. source
| rm -f .tmp_quiet_recordmcount
| /bin/bash ../scripts/mkmakefile \
|     .. . 2016 07
|   GEN     ./Makefile
| make -f ../scripts/Makefile.build obj=scripts/kconfig silentoldconfig
[... 51 lines cut ...]
| ../arch/arm/mach-aspeed/platform_g4.S: Assembler messages:
| ../arch/arm/mach-aspeed/platform_g4.S:681: Error: bad expression -- `ldr r2,=#0xffffffef'
| ../arch/arm/mach-aspeed/platform_g4.S:681: Error: bad expression -- `ldr r2,=#0xffffffdf'
| ../scripts/Makefile.build:316: recipe for target 'arch/arm/mach-aspeed/platform_g4.o' failed
| make[2]: *** [arch/arm/mach-aspeed/platform_g4.o] Error 1
[... cut ...]
ERROR: Task 6 (/home/parallels/openBmc/poky/meta-openbmc/meta-aspeed/recipes-bsp/u-boot/u-boot_2016.07.bb, do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 371 tasks of which 370 didn't need to be rerun and 1 failed.
No currently running tasks (349 of 379)

Summary: 1 task failed:
  /home/parallels/openBmc/poky/meta-openbmc/meta-aspeed/recipes-bsp/u-boot/u-boot_2016.07.bb, do_compile
Summary: There was 1 WARNING message shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
parallels@ubuntu:~/openBmc/poky/build$

AdrianB-Flex, 2017-05-10:
I am getting a similar error although I thought Ubuntu 14.04 was the latest to be supported for openbmc builds.

FrankHome, 2017-05-10:
HI,AdrianB-Flex 
Did you succeed Finally？

AdrianB-Flex, 2017-05-10:
No.  Still pulling my hair out.  I followed exactly what was on the "How to build" page but no joy yet.  I am working with someone in Issue #47 so hopefully soon.

FrankHome, 2017-05-11:
@AdrianB-Flex 
Waiting for your good news.and i will keep trying

closed 2017-05-13 by FrankHome

theopolis, 2017-05-21:
Hey, @FrankHome, @AdrianB-Flex was able to solve the build issue, check out the discussion in #47.