facebook/openbmc #50: Question about defining UBOOT_MACHINE names

AdrianB-Flex, 2017-05-19:
I have a new machine and in meta-newmachine/meta-foo/conf/machine/foo.conf I defined 
UBOOT_MACHINE_foo= "foo_config"
but this will not compile.  Fails in u-boot_2016.07.bb (or possibly u-boot-fw-utils_2016.07.bb, not sure which.)
If I rename it to any of the facebook configs (i.e. fbwedge_config) it compiles fine which tells me there is some further configuration I am not doing.

Would appreciate any help with this, in particular to whatever documentation I should be referencing.

theopolis, 2017-05-21:
This is because Yocto, the U-boot recipes, will look for a `./configs/NAME_defconfig` file in the U-Boot source code. If that does not exist it will cause the failure.

You'll need to tweak some of the U-Boot recipe logic to develop for a new board, since the recipe: https://github.com/facebook/openbmc/blob/helium/meta-aspeed/recipes-bsp/u-boot/u-boot_2016.07.bb#L11 is building U-Boot from another repo.

To do that the general guidance is to change the logic to use a local `./files/u-boot-v2016.07` folder containing the source. Then add your new machine configuration/etc. This will be fairly challenging and it's recommended you have a bit of experience with Yocto and U-Boot building.

AdrianB-Flex, 2017-05-22:
Thank you Theopolis.  I have now read through the u-boot README and am getting a clearer picture.  Good starting place for some experimentation.  Thanks.

closed 2017-05-22 by AdrianB-Flex