facebook/openbmc #52: Is it possible to "freeze" my local source?

AdrianB-Flex, 2017-06-05:
One of my bosses wants our complete openbmc implementation frozen and archived locally.  I have explained I don't think that is possible since many repos are pulled during the build process (not to mention the syncing nightmare that would eventually lead to).  But I understand the concern, how would I maintain the code for our product if the source can and will change?

tfangit, 2017-06-05:
In meta-facebook/conf/distro/openbmc-fb.conf, we defined 'BB_GENERATE_MIRROR_TARBALLS = "1"'. That tells bitbake to generate tarball for all source code downloaded for the build. You can then copy those tarballs into another directory as source mirror and track it under version control. Then modify "SOURCE_MIRROR_URL" to point to the source mirror repo. After that, you will be able to generate your build completely based on your local source.

AdrianB-Flex, 2017-06-05:
That's good to know.  Thanks Tian.

AdrianB-Flex, 2017-06-05:
Where are the tarballs saved?

Legal Disclaimer:
The information contained in this message may be privileged and confidential. It is intended to be read only by the individual or entity to whom it is addressed or by their designee. If the reader of this message is not the intended recipient, you are on notice that any distribution of this message, in any form, is strictly prohibited. If you have received this message in error, please immediately notify the sender and delete or destroy any copy of this message!

AdrianB-Flex, 2017-06-08:
Tian, where can I find the tarball?

closed 2018-04-04 by AdrianB-Flex