facebook/openbmc #56: Build fail in obmc-pal

UlyssesKao0801, 2017-07-07:
try to use config TEMPLATECONF=meta-openbmc/meta-facebook/meta-fbtp/conf to build image.
I meet the build error 

[ 50%] Building C object CMakeFiles/obmc-pal.dir/obmc-pal.c.o
| /home/ukao/openbmc/poky/build/tmp/sysroots/x86_64-linux/usr/bin/arm-fb-linux-gnueabi/arm-fb-linux-gnueabi-gcc    -march=armv6  --sysroot=/home/ukao/openbmc/poky/build/tmp/sysroots/fbtp  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/ukao/openbmc/poky/build/tmp/work/armv6-fb-linux-gnueabi/obmc-pal/0.1-r1=/usr/src/debug/obmc-pal/0.1-r1 -fdebug-prefix-map=/home/ukao/openbmc/poky/build/tmp/sysroots/x86_64-linux= -fdebug-prefix-map=/home/ukao/openbmc/poky/build/tmp/sysroots/fbtp=    -o CMakeFiles/obmc-pal.dir/obmc-pal.c.o   -c /home/ukao/openbmc/poky/build/tmp/work/armv6-fb-linux-gnueabi/obmc-pal/0.1-r1/obmc-pal.c
| /home/ukao/openbmc/poky/build/tmp/work/armv6-fb-linux-gnueabi/obmc-pal/0.1-r1/obmc-pal.c:30:24: fatal error: openbmc/kv.h: No such file or directory
| compilation terminated.

Please help to resolve this issue

vrpradeepkumar, 2017-08-17:
Hi ,
It works fine for me.
Can you be more specific?

I used the following steps to build 

export TEMPLATECONF=meta-openbmc/meta-facebook/meta-fbtp/conf
. oe-init-build-env
bitbake fbtp-image

Thanks

henbin, 2017-08-17:
@vrpradeepkumar Thanks for your response. I have passed this issue.

UlyssesKao0801, 2017-08-21:
Dear all

it OK for me now too

________________________________
寄件者: henbin <notifications@github.com>
寄件日期: 2017年8月17日 下午 03:33
收件者: facebook/openbmc
副本: UlyssesKao0801; Author
主旨: Re: [facebook/openbmc] Build fail in obmc-pal (#56)

@vrpradeepkumar<https://github.com/vrpradeepkumar> Thanks for your response. I have passed this issue.