facebook/openbmc #57: is there pldm and ncsi support available?

mb2rohit, 2017-07-09:
i looked at the code and find only mctp over i2c support available, but README doesn't say any thing about its availability, i just wanted to know if it is working and also if PLDM over MCTP, NCSI over MCTP support is also available?

gmissonnier, 2018-01-04:
+1

I found this small MCTP example :
./meta-facebook/meta-fbttn/recipes-fbttn/fblibs/files/mctp/mctp.c

Is this example known to respect the MCTP standard?

From my understanding, MCTP over I2C requires to use only SMBUS Block Write protocol, with PEC enabled, and it need both side to have a slave address to enable bi-directionnal communications.
The example does not seems to receive MCTP response this way, and I do not see the PEC byte in the requests packets!

merline666, 2019-04-25:
[image]

merline666, 2019-04-25:
<p>[image]</p>

merline666, 2019-04-25:
nice <p>[image]</p>