facebook/openbmc #62: Error while building the image after changing the patch file

radhikapradeep, 2017-08-28:
Hi ,

We are changing some value in the patch file "https://github.com/facebook/openbmc/blob/master/meta-aspeed/recipes-kernel/linux/files/patch-2.6.28.9/0000-linux-aspeed-064.patch" for baud rate calculation and building the image for our wedge100 like platform.

But it is showing Hunk failure error everytime.

When checked , the change in patch file is reflected in the files in the build directory.

Is there some additional steps to be done to change the patch file?

Thanks and Regards

covracer, 2017-09-25:
Rather than editing the patch file, I would recommend creating a new patch file using `diff -u`: https://stackoverflow.com/questions/437219/using-the-output-of-diff-to-create-the-patch/437241#437241

radhikapradeep, 2017-10-30:
Ok Thanks.

closed 2017-10-30 by radhikapradeep