facebook/openbmc #70: Accessing Wedge100 EEPROM

Ananth-CG, 2017-10-05:
In the **wedge_eeprom.c** file, I2C bus - 6 with address 0x50 is being accessed to obtain board info. Which is this EEPROM ? There is no EEPROM with address 0x50 connected to this bus as referred from schematics and block diagram. As I understand, there is one Additional EEPROM with address 0x50, but it is controlled by micro-server, not wedge100 BMC.

I tried **_i2cdetect -a 6_** in the console and the response is as follows:

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 3f
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- UU -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --

As I understand, this **wedge_eeprom** seems to be looking for Board related information like Product name, version, serial number, MAC and so on. And these are accessed mainly by fruid.c in **IPMID** and **OOBNIC**'s main.c. The issue here is that there is an error log observed in the console as follows:

_ipmid: populate_fruid: wedge_eeprom_parse returns -14_

What could be the reason for this ?
Thanks in advance.

jmehta, 2017-11-14:
Hello @Ananth-CG, sorry for the delayed response. 

Usually the address for any i2c device in the schematic is 8-bit based vs the same in linux kernel is 7-bit based. 
So, looking at your `i2cdetect -a 6` output, the eeprom seems to be on address 0x51 (7-bit based) which is `0x51 << 1 = 0xA2 (8-bit based)` in the platform schematic. 

Similarly the i2c_bus numbering is different in schematic (1-based) and in linux kernel (0-based). So, the device in EEPROM device in the wedge schematic should be on  `i2c_bus 7; addr 0xA2`. 

Please let me know if you have more questions.

Ananth-CG, 2017-11-27:
Hi @jmehta ,
Thanks for the reply.
According to the schematics, the I2C bus is I2C7 for this EEPROM. Moreover, as per wedge_eeprom.c file in "poky/meta-openbmc/meta-facebook/recipes-utils/wedge-eeprom/files/lib/", the I2C path is "/sys/class/i2c-adapter/i2c-6/6-0050/eeprom", which means I2C bus 6, address 0x50.

I think this was the issue. There is a "libwedge-eeprom_0.1.bbappend" file in "poky/meta-openbmc/meta-facebook/meta-wedge100/recipes-utils/wedge-eeprom/", according to which the path is "/sys/class/i2c-adapter/i2c-6/6-0051/eeprom", which means I2C bus 6, address 0x51. When I changed the bus address path in wedge_eeprom.c file from 6-0050 to 6-0051, the error was resolved.

Just wanted to know, why is it different ?