facebook/openbmc #76: BMC is not reachable always.

anwibrah, 2017-12-20:


anwibrah, 2017-12-20:
some  logs.
ping from my mac

ANWIBRAH-M-629A:wedge anwibrah$ ping 172.31.207.125
PING 172.31.207.125 (172.31.207.125): 56 data bytes
64 bytes from 172.31.207.125: icmp_seq=0 ttl=58 time=4.714 ms
64 bytes from 172.31.207.125: icmp_seq=1 ttl=58 time=4.153 ms
64 bytes from 172.31.207.125: icmp_seq=2 ttl=58 time=5.699 ms
64 bytes from 172.31.207.125: icmp_seq=3 ttl=58 time=8.307 ms
64 bytes from 172.31.207.125: icmp_seq=4 ttl=58 time=5.058 ms
64 bytes from 172.31.207.125: icmp_seq=5 ttl=58 time=3.628 ms
64 bytes from 172.31.207.125: icmp_seq=6 ttl=58 time=4.970 ms
64 bytes from 172.31.207.125: icmp_seq=7 ttl=58 time=4.413 ms
64 bytes from 172.31.207.125: icmp_seq=8 ttl=58 time=103.264 ms
64 bytes from 172.31.207.125: icmp_seq=9 ttl=58 time=4.233 ms
64 bytes from 172.31.207.125: icmp_seq=10 ttl=58 time=1.634 ms
64 bytes from 172.31.207.125: icmp_seq=11 ttl=58 time=2.070 ms
64 bytes from 172.31.207.125: icmp_seq=12 ttl=58 time=1.558 ms
64 bytes from 172.31.207.125: icmp_seq=13 ttl=58 time=3.981 ms
64 bytes from 172.31.207.125: icmp_seq=14 ttl=58 time=1.467 ms
64 bytes from 172.31.207.125: icmp_seq=15 ttl=58 time=4.371 ms
64 bytes from 172.31.207.125: icmp_seq=16 ttl=58 time=4.515 ms
64 bytes from 172.31.207.125: icmp_seq=17 ttl=58 time=1.875 ms
64 bytes from 172.31.207.125: icmp_seq=18 ttl=58 time=1.244 ms
64 bytes from 172.31.207.125: icmp_seq=19 ttl=58 time=2.435 ms
64 bytes from 172.31.207.125: icmp_seq=20 ttl=58 time=1.539 ms
64 bytes from 172.31.207.125: icmp_seq=21 ttl=58 time=4.448 ms
64 bytes from 172.31.207.125: icmp_seq=22 ttl=58 time=4.915 ms
64 bytes from 172.31.207.125: icmp_seq=23 ttl=58 time=1.673 ms
64 bytes from 172.31.207.125: icmp_seq=24 ttl=58 time=4.535 ms
64 bytes from 172.31.207.125: icmp_seq=25 ttl=58 time=4.509 ms
64 bytes from 172.31.207.125: icmp_seq=26 ttl=58 time=3.952 ms
64 bytes from 172.31.207.125: icmp_seq=27 ttl=58 time=34.443 ms
64 bytes from 172.31.207.125: icmp_seq=28 ttl=58 time=2.224 ms
64 bytes from 172.31.207.125: icmp_seq=29 ttl=58 time=5.324 ms
64 bytes from 172.31.207.125: icmp_seq=30 ttl=58 time=4.715 ms
64 bytes from 172.31.207.125: icmp_seq=31 ttl=58 time=6.265 ms
64 bytes from 172.31.207.125: icmp_seq=32 ttl=58 time=4.294 ms
64 bytes from 172.31.207.125: icmp_seq=33 ttl=58 time=6.293 ms
Request timeout for icmp_seq 34 ---> broken
Request timeout for icmp_seq 35
Request timeout for icmp_seq 36
Request timeout for icmp_seq 37
Request timeout for icmp_seq 38
Request timeout for icmp_seq 39
Request timeout for icmp_seq 40
Request timeout for icmp_seq 41
Request timeout for icmp_seq 42
Request timeout for icmp_seq 43
Request timeout for icmp_seq 44
Request timeout for icmp_seq 45
Request timeout for icmp_seq 46
Request timeout for icmp_seq 47
Request timeout for icmp_seq 48
Request timeout for icmp_seq 49
Request timeout for icmp_seq 50
Request timeout for icmp_seq 51
Request timeout for icmp_seq 52
Request timeout for icmp_seq 53
Request timeout for icmp_seq 54
Request timeout for icmp_seq 55
Request timeout for icmp_seq 56
Request timeout for icmp_seq 57
Request timeout for icmp_seq 58
Request timeout for icmp_seq 59
Request timeout for icmp_seq 60
Request timeout for icmp_seq 61
Request timeout for icmp_seq 62
64 bytes from 172.31.207.125: icmp_seq=63 ttl=58 time=2.551 ms
64 bytes from 172.31.207.125: icmp_seq=64 ttl=58 time=4.549 ms
[... 27 lines cut ...]
64 bytes from 172.31.207.125: icmp_seq=92 ttl=58 time=4.686 ms
64 bytes from 172.31.207.125: icmp_seq=93 ttl=58 time=2.821 ms
Request timeout for icmp_seq 94 --> broken
Request timeout for icmp_seq 95
Request timeout for icmp_seq 96
Request timeout for icmp_seq 97
Request timeout for icmp_seq 98
Request timeout for icmp_seq 99
Request timeout for icmp_seq 100
Request timeout for icmp_seq 101
Request timeout for icmp_seq 102
Request timeout for icmp_seq 103
Request timeout for icmp_seq 104
Request timeout for icmp_seq 105
64 bytes from 172.31.207.125: icmp_seq=106 ttl=58 time=4.487 ms
64 bytes from 172.31.207.125: icmp_seq=107 ttl=58 time=4.794 ms
64 bytes from 172.31.207.125: icmp_seq=108 ttl=58 time=12.207 ms
64 bytes from 172.31.207.125: icmp_seq=109 ttl=58 time=2.052 ms
64 bytes from 172.31.207.125: icmp_seq=110 ttl=58 time=1.365 ms
64 bytes from 172.31.207.125: icmp_seq=111 ttl=58 time=4.366 ms
64 bytes from 172.31.207.125: icmp_seq=112 ttl=58 time=4.196 ms
64 bytes from 172.31.207.125: icmp_seq=113 ttl=58 time=4.286 ms
64 bytes from 172.31.207.125: icmp_seq=114 ttl=58 time=4.464 ms
64 bytes from 172.31.207.125: icmp_seq=115 ttl=58 time=5.000 ms
64 bytes from

covracer, 2017-12-27:
Does this reproduce on the 4.1 kernel in a build from the Helium branch?