facebook/openbmc #80: Flooding on eth0 causes reboot

AdrianB-Flex, 2018-03-12:
Our board has an additional GigE module that under certain conditions is flooding the eth0 interface on the BMC.  When this happens it causes the BMC to endlessly reboot.  We have found a way to mitigate this from the GigE switch side but it is cumbersome.  Is there anyway I could configure the ethernet driver on the BMC to drop these pkts or aply backpressure or something?  Our openbmc implementation is based on code frozen about a year ago.  Linux kernel is 2.6.28.9

williamspatrick, 2018-03-12:
There was a fairly extensive rewrite of the ftgmac driver in the Linux kernel last year roughly in April.  I'm not sure how well these would be able to be backported to 2.6.28.9, but they resulted in a significant performance improvement.

https://github.com/torvalds/linux/commits/5bbcc0f595fadb4cac0eddc4401035ec0bd95b09/drivers/net/ethernet/faraday/ftgmac100.c

AdrianB-Flex, 2018-03-12:
I was *afraid* of that.  Guess it might make more sense looking to upgrade to the latest BMC.  Thanks Patrick.

closed 2018-03-12 by AdrianB-Flex