facebook/openbmc #83: REST calls failing from COMe (microserver)

AdrianB-Flex, 2018-04-25:
Have been using an older frozen repo of openbmc where everything was working for me (linux2.6.x).  Just moved to the latest helium and I am unable to make REST calls from the COMe - I am getting a connection refused.  I have no local rest changes and I am using the eth0 interface.
From the COMe:
-bash-4.2$ sudo ping 10.115.16.50
[sudo] password for admin:
PING 10.115.16.50 (10.115.16.50) 56(84) bytes of data.
64 bytes from 10.115.16.50: icmp_seq=1 ttl=64 time=0.416 ms
64 bytes from 10.115.16.50: icmp_seq=2 ttl=64 time=0.438 ms
64 bytes from 10.115.16.50: icmp_seq=3 ttl=64 time=0.332 ms
64 bytes from 10.115.16.50: icmp_seq=4 ttl=64 time=0.414 ms

--- 10.115.16.50 ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 2997ms
rtt min/avg/max/mdev = 0.332/0.400/0.438/0.040 ms

-bash-4.2$ sudo curl -i -H "Accept: application/json" http://10.115.16.50:8080/api/sys/platform
[sudo] password for admin:
curl: (7) Failed connect to 10.115.16.50:8080; Connection refused

From BMC:
root@bmc-oob:~# uname -a
Linux bmc-oob 4.1.15-bolt #1 Mon Apr 23 20:51:20 PDT 2018 armv5tejl GNU/Linux
root@bmc-oob:~#

When I look at the tasks running I see the expected:
  729 root         0 Z    [run]
  730 root      2844 S    runsv /etc/sv/restapi
  731 root      2844 S    runsv /etc/sv/restwatchdog
  734 root     74016 S    python3 /usr/local/bin/rest.py
  736 root     16472 S    python3 /usr/local/bin/rest_watchdog.py

AdrianB-Flex, 2018-04-26:
Switched back to using the USB interface to eliminate any network issue but usb0 is not coming up.  USB is the interface my code on the microserver is using to query the BMC.  I see this on bootup:   "IPv6: ADDRCONF(NETDEV_UP): usb0: link is not ready".  Has anyone else seen this?  What would prevent usb0 from getting a link?

AdrianB-Flex, 2018-05-15:
Fixed the usb0 issue and am now using that internal interface but REST calls are still being refused.

sdasari73, 2018-09-05:
Any clues from /tmp/rest.log file?

AdrianB-Flex, 2018-09-11:
Solved a couple months ago

closed 2018-09-11 by AdrianB-Flex