facebook/openbmc #86: ssh is dual stack but not http server? (wedge100 Helium)

AdrianB-Flex, 2018-06-11:
root@bmc-oob:/etc# netstat -lna | grep LISTEN
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN
tcp        0      0 :::8080                 :::*                    LISTEN
tcp        0      0 :::22                   :::*                    LISTEN
unix  2      [ ACC ]     STREAM     LISTENING       6403 /var/run/rackmond.sock
unix  2      [ ACC ]     SEQPACKET  LISTENING       3257 /run/udev/control
unix  2      [ ACC ]     STREAM     LISTENING       6363 /tmp/ipmi_socket
root@bmc-oob:/etc#
root@bmc-oob:/etc# uname -a
Linux bmc-oob 4.1.51-wedge100 #1 Fri May 25 13:22:12 PDT 2018 armv5tejl GNU/Linux
root@bmc-oob:/etc#

sdasari73, 2018-09-05:
Does the system acquired both IPv4 and IPv6 addresses? Can you try sending REST API commands to both IP addresses and check the response?

AdrianB-Flex, 2018-09-11:
Solved a couple months ago.

closed 2018-09-11 by AdrianB-Flex