facebook/openbmc #95: Question on baud rate of sol-util / mTerm on Tioga Pass

MennoCB, 2018-10-21:
Hi,

Facebook 2S Server Tioga Pass specification says:
Setting for the local physical COM port (COM0) and SOL (COM1): The default is enable console redirection on both ports with baud rate 57600, no flow control, terminal type VT100, 8 data bits, No Parity, 1 Stop Bit."
But we have noticed that our Tioga Pass channel version motherboards come with a default of 115200 for the BIOS, which is probably the default setting from AMI.

With everything set to 57600 it works just fine. But I have found no setting for baud rate in OpenBMC, is it possible to set it to something else besides 57600?
I don't see this as an OpenBMC issues, but it would be nice in those cases where your devices turns out to have a different setting.

Thanks & Regards,
Menno

amithash, 2018-10-22:
The BMC console baudrate is set in different components depending on who has control of the UART during boot.
Uboot, kernel:
CONFIG_BAUDRATE and CONFIG_BOOTARGS in include/configs/{platform}.h.
Example: include/configs/fbtp.h
Once user-space takes over, it is in common/recipes-utils/openbmc-utils/files/revise_inittab and meta-facebook/meta-{platform}/recipes-core/sysvinit/sysvinit-inittab_%.bbappend

The SOL (host-server) baud-rate is not under BMC control. So, I would recommend sticking to 57600.

MennoCB, 2018-10-23:
My question was not so much about the baud rate of the BMC, but what if the host-server baud rate is not set to 57600, is there any option to change it on the SOL client side?
Of course recommended is 57600, but in my case the BIOS was set to 115200 by default, it could also be someone that installs the OS and sets it to 115200 by mistake. I would still like to be able to reach that BIOS or OS through SOL...

tfangit, 2018-10-23:
The current mTerm hard code the baudrate to 57600. In order to change it, you will need to modify the code and build/flash.

It won't be difficult to add an option to mTerm to accept baudrate parameter and then use board layer to override it.

One more enhancement could be to allow mTerm client to send a cmd to mTerm server to change the baudrate on the fly.

amithash, 2018-10-23:
@MennoCB Thanks for the clarification! Re-opening the issue. Lets use this issue to track runtime configuration of baud rate as proposed by @tfangit. 
In the meanwhile, you might need to recompile after changing tty_helper.h:#define BAUDRATE B57600

MennoCB, 2018-10-23:
Thank you Amit, that is great and this already helps me with all the channel version gear from this manufacturer.