facebook/openbmc #100: meta-portwell is broken

noamc, 2019-04-05:
Hello
Any ideas why meta-protwell is broken even before I change anything?

(helium) ncamus@ncamus-VM:openbmc $ source openbmc-init-build-env meta-portwell/ build-portwell
classes/          conf/             COPYING           meta-pwneptune/   meta-python3/     recipes-core/     recipes-extended/ recipes-support/  recipes-utils/    
(helium) ncamus@ncamus-VM:openbmc $ source openbmc-init-build-env meta-portwell/meta-pwneptune build-portwell
WARNING: distro is set to rocko based on DISTRO_DEFAULT env variable
Init build environment for platform 'meta-portwell/meta-pwneptune' with distro rocko...
You had no conf/local.conf file. This configuration file has therefore been
created for you with some default values. You may wish to edit it to, for
example, select a different MACHINE (target hardware). See conf/local.conf
for more information as common configuration options are commented.

You had no conf/bblayers.conf file. This configuration file has therefore been
created for you with some default values. To add additional metadata layers
into your configuration please add entries to conf/bblayers.conf.

The Yocto Project has extensive documentation about OE including a reference
manual which can be found at:
    http://yoctoproject.org/documentation

For more information about OpenEmbedded see their website:
    http://www.openembedded.org/

Common targets are:
    pwneptune-image
(helium) ncamus@ncamus-VM:build-portwell $ bitbake pwneptune-image
NOTE: Your conf/bblayers.conf has been automatically updated.
WARNING: Host distribution "ubuntu-18.04" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
WARNING: /home/ncamus/trees/npu-stack/embedded/openbmc/yocto/rocko/poky/../../../meta-aspeed/recipes-kernel/linux/linux-aspeed_4.17.bb: You are using the linux-dtb.inc which is deprecated. You can safely remove it as the Device Tree support is automatically enabled when KERNEL_DEVICETREE is set.
WARNING: /home/ncamus/trees/npu-stack/embedded/openbmc/yocto/rocko/poky/../../../meta-aspeed/recipes-kernel/linux/linux-aspeed_4.18.bb: You are using the linux-dtb.inc which is deprecated. You can safely remove it as the Device Tree support is automatically enabled when KERNEL_DEVICETREE is set.
WARNING: /home/ncamus/trees/npu-stack/embedded/openbmc/yocto/rocko/poky/../../../meta-aspeed/recipes-kernel/linux/linux-aspeed_4.1.bb: You are using the linux-dtb.inc which is deprecated. You can safely remove it as the Device Tree support is automatically enabled when KERNEL_DEVICETREE is set.
Parsing recipes: 100% |###########################################################################################################################################################################| Time: 0:01:28
Parsing of 2058 .bb files complete (0 cached, 2058 parsed). 2862 targets, 315 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'update-rc.d-nativelibipmi' (but /home/ncamus/trees/npu-stack/embedded/openbmc/yocto/rocko/poky/../../../common/recipes-core/ipmid/ipmid_0.2.bb DEPENDS on or otherwise requires it). Close matches:
  update-rc.d-native
NOTE: Runtime target 'ipmid' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['ipmid', 'update-rc.d-nativelibipmi']
ERROR: Required build target 'pwneptune-image' has no buildable providers.
Missing or unbuildable dependency chain was: ['pwneptune-image', 'ipmid', 'update-rc.d-nativelibipmi']

Summary: There were 4 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

benwei13, 2019-04-11:
Is this still broken?  I tried it just now and pwneptune builds okay at the moment

noamc, 2019-04-12:
Yes it is, I see this:
(helium) ncamus@ncamus-VM:openbmc $ git show --oneline 
b47662607 (HEAD -> helium, upstream/helium, origin/helium) fbtp: Make the BMC JTAG controller become the slave after asd test
(helium) ncamus@ncamus-VM:openbmc $ ./sync_yocto.sh 
Cloning into 'yocto/krogoth/poky'...
remote: Counting objects: 437185, done.
remote: Compressing objects: 100% (103279/103279), done.
remote: Total 437185 (delta 327015), reused 436908 (delta 326775)
Receiving objects: 100% (437185/437185), 154.82 MiB | 6.66 MiB/s, done.
Resolving deltas: 100% (327015/327015), done.
Cloning into 'yocto/krogoth/meta-openembedded'...
remote: Enumerating objects: 199, done.
remote: Counting objects: 100% (199/199), done.
remote: Compressing objects: 100% (111/111), done.
remote: Total 106607 (delta 102), reused 154 (delta 85), pack-reused 106408
Receiving objects: 100% (106607/106607), 36.76 MiB | 11.63 MiB/s, done.
Resolving deltas: 100% (61557/61557), done.
Cloning into 'yocto/krogoth/meta-security'...
remote: Counting objects: 3409, done.
remote: Compressing objects: 100% (1485/1485), done.
remote: Total 3409 (delta 1781), reused 3329 (delta 1706)
Receiving objects: 100% (3409/3409), 3.44 MiB | 525.00 KiB/s, done.
Resolving deltas: 100% (1781/1781), done.
Cloning into 'yocto/rocko/poky'...
remote: Counting objects: 437185, done.
remote: Compressing objects: 100% (103279/103279), done.
remote: Total 437185 (delta 327015), reused 436908 (delta 326775)
Receiving objects: 100% (437185/437185), 154.82 MiB | 6.54 MiB/s, done.
Resolving deltas: 100% (327015/327015), done.
Cloning into 'yocto/rocko/meta-openembedded'...
remote: Enumerating objects: 199, done.
remote: Counting objects: 100% (199/199), done.
remote: Compressing objects: 100% (111/111), done.
remote: Total 106607 (delta 102), reused 154 (delta 85), pack-reused 106408
Receiving objects: 100% (106607/106607), 36.76 MiB | 11.12 MiB/s, done.
Resolving deltas: 100% (61557/61557), done.
Cloning into 'yocto/rocko/meta-security'...
remote: Counting objects: 3409, done.
remote: Compressing objects: 100% (1485/1485), done.
remote: Total 3409 (delta 1782), reused 3328 (delta 1706)
Receiving objects: 100% (3409/3409), 3.44 MiB | 367.00 KiB/s, done.
Resolving deltas: 100% (1782/1782), done.
(helium) ncamus@ncamus-VM:openbmc $ 
(helium) ncamus@ncamus-VM:openbmc $ source openbmc-init-build-env meta-portwell/
classes/          COPYING           meta-python3/     recipes-extended/ recipes-utils/    
conf/             meta-pwneptune/   recipes-core/     recipes-support/  
(helium) ncamus@ncamus-VM:openbmc $ source openbmc-init-build-env meta-portwell/meta-pwneptune/ build-portwell
Init build environment for platform 'meta-portwell/meta-pwneptune/' with distro rocko...
You had no conf/local.conf file. This configuration file has therefore been
created for you with some default values. You may wish to edit it to, for
example, select a different MACHINE (target hardware). See conf/local.conf
for more information as common configuration options are commented.

You had no conf/bblayers.conf file. This configuration file has therefore been
created for you with some default values. To add additional metadata layers
into your configuration please add entries to conf/bblayers.conf.

The Yocto Project has extensive documentation about OE including a reference
manual which can be found at:
    http://yoctoproject.org/documentation

For more information about OpenEmbedded see their website:
    http://www.openembedded.org/

Common targets are:
    pwneptune-image
(helium) ncamus@ncamus-VM:build-portwell $ bitbake pwneptune-image
NOTE: Your conf/bblayers.conf has been automatically updated.
WARNING: Host distribution "ubuntu-18.04" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
WARNING: /home/ncamus/trees/npu-stack/embedded/openbmc/yocto/rocko/poky/../../../meta-aspeed/recipes-kernel/linux/linux-aspeed_5.0.bb: You are using the linux-dtb.inc which is deprecated. You can safely remove it as the Device Tree support is automatically enabled when KERNEL_DEVICETREE is set.
[... 64 lines cut ...]
ERROR: Task (virtual:native:/home/ncamus/trees/npu-stack/embedded/openbmc/yocto/rocko/poky/meta/recipes-devtools/automake/automake_1.15.1.bb:do_fetch) failed with exit code '1'
ERROR: autoconf-native-2.69-r11 do_fetch: The URL: 'http://ftp.gnu.org/gnu/autoconf/autoconf-2.69.tar.gz' is not trusted and cannot be used
ERROR: autoconf-native-2.69-r11 do_fetch: Fetcher failure for URL: 'http://ftp.gnu.org/gnu/autoconf/autoconf-2.69.tar.gz'. Unable to fetch URL from any source.
ERROR: autoconf-native-2.69-r11 do_fetch: Function failed: base_do_fetch
ERROR: Logfile of failure stored in: /home/ncamus/trees/npu-stack/embedded/openbmc/build-portwell/tmp/work/x86_64-linux/autoconf-native/2.69-r11/temp/log.do_fetch.5768
ERROR: Task (virtual:native:/home/ncamus/trees/npu-stack/embedded/openbmc/yocto/rocko/poky/meta/recipes-devtools/autoconf/autoconf_2.69.bb:do_fetch) failed with exit code '1'
WARNING: xz-native-5.2.3-r0 do_fetch: Failed to fetch URL http://tukaani.org/xz/xz-5.2.3.tar.gz, attempting MIRRORS if available
ERROR: xz-native-5.2.3-r0 do_fetch: The URL: 'http://tukaani.org/xz/xz-5.2.3.tar.gz' is not trusted and cannot be used
ERROR: xz-native-5.2.3-r0 do_fetch: Fetcher failure for URL: 'http://tukaani.org/xz/xz-5.2.3.tar.gz'. Unable to fetch URL from any source.
[... cut ...]
  virtual:native:/home/ncamus/trees/npu-stack/embedded/openbmc/yocto/rocko/poky/meta/recipes-devtools/automake/automake_1.15.1.bb:do_fetch
  virtual:native:/home/ncamus/trees/npu-stack/embedded/openbmc/yocto/rocko/poky/meta/recipes-devtools/autoconf/autoconf_2.69.bb:do_fetch
  virtual:native:/home/ncamus/trees/npu-stack/embedded/openbmc/yocto/rocko/poky/meta/recipes-extended/xz/xz_5.2.3.bb:do_fetch
Summary: There were 12 WARNING messages shown.
Summary: There were 21 ERROR messages shown, returning a non-zero exit code.
(helium) ncamus@ncamus-VM:build-portwell $

noamc, 2019-04-27:
Seem like that if BB_ALLOWED_NETWORKS can be defined properly above do_fetch failures will be resolved (no more "is not trusted and cannot be used").

Can someone tell me in what file should I add such definition so bitbake will use it?

noamc, 2019-04-27:
OK found this in meta-portwell/conf/distro/openbmc-fb.conf:
# The following setting will prevent bitbake from downloading anything over the
# network.  This can be used to ensure that we get everything from a local
# file:// mirror.
#
# Comment this out if you do need to download new packages from the internet.
# However, once you have downloaded the package you should check them into our
# mirror repository so that other developers will always get it from the mirror
# repo.
INHERIT += "fb-only-network"

closed 2019-05-25 by noamc