facebook/openbmc #118: How to build OpenBMC for AST2600

rfeldman94303, 2020-04-17:
Hi All,

I am trying to build a functional OpenBMC AST2600 image, in order to run sanity tests on our evaluation AST2600 board. Unfortunately, I have not been able to find clear instructions, under https://github.com/openbmc/openbmc, pertaining to how to build an OpenBMC AST2600 image...

Hence, I have the following questions for the OpenBMC team:
1. Has anyone been able to successfully build an OpenBMC AST2600 image? If yes:
 - Please provide a link to instructions for how to build this image.
 - Is it currently possible to run sanity tests on this OpenBMC AST2600 image using QEMU? If yes, it would be great if you could provide a link to corresponding unit test logs.
 - Has this image been successfully installed on actual AST2600 hardware? If yes, it would be great if you can provide a link to corresponding unit test logs.
  - Is there a particular OpenBMC branch/release which is considered to be the most stable to building and testing AST2600 images?

Thanks,
Robert
robert.feldman@paviliondata.io

williamspatrick, 2020-05-21:
@rfeldman94303 I don't think we have any AST2600-based systems in our tree at this point.

I'm pretty certain that the Linux Foundation OpenBMC does have some AST2600 work done on the Aspeed SDK, but I don't think they have any AST2600-based systems yet either.  @shenki has been doing kernel work and @legoater has done qemu work for AST2600 over there.

Best suggestion to get feedback from them is to post to the openbmc mailing list.  These github issues tend to get lost.

closed 2020-05-21 by williamspatrick

rfeldman94303, 2020-05-21:
Thanks,
Robert

apoorvemohan, 2020-12-09:
@rfeldman94303 Were you able to build/get OpenBMC AST2600 image for QEMU? If yes, could you please share some docs on how to re-produce it? Thanks!