facebook/openbmc #124: Support aspeed 2600

dineshkumarv86, 2020-07-24:
Hi all, 
I am newbie , in our future project we are looking to add ast2600 bmc chip. Is it support for ast2600 BMC chip is available in facebook/openbmc ?

williamspatrick, 2020-07-27:
@dineshkumarv86 - We use the linux foundation openbmc kernel, which supports ast2600.  We also have 3 machines we are working on development of that are using the ast2600 hardware.

```
meta-facebook/meta-cloudripper/conf/machine/cloudripper.conf:require conf/machine/include/ast2620.inc
meta-facebook/meta-fuji/conf/machine/fuji.conf:require conf/machine/include/ast2620.inc
meta-facebook/meta-grandcanyon/conf/machine/grandcanyon.conf:require conf/machine/include/ast2620.inc
```

closed 2020-07-27 by williamspatrick

dagilson-ciena, 2020-08-25:
I don't see cloudripper-image, fuji-image or grandcanyon-image targets.
I have an AST2600 eval board coming in and would like to know how to properly build one of these images. 

I tried:
source openbmc-init-build-env meta-facebook/meta-fuji
but 
bitbake fuji-image
.
.
WARNING: No recipes available for:
  /localdata/dagilson/openbmc/yocto/warrior/poky/../../../common/recipes-tpm2/tpm2-tools/tpm2-tools_3.2.0.bbappend
ERROR: Nothing PROVIDES 'fuji-image'

williamspatrick, 2020-10-20:
@dagilson-ciena - Have you been able to resolve this?  I don't see any obvious reason why this would fail.

`source openbmc-init-build-env fuji` should be sufficient (not the whole path to the meta layer).

Have you ran the 'sync_yocto.sh`?