facebook/openbmc #133: AST2600 Support

dagilson-ciena, 2020-09-01:
From issue 124

I see the includes below:

meta-facebook/meta-cloudripper/conf/machine/cloudripper.conf:require conf/machine/include/ast2620.inc
meta-facebook/meta-fuji/conf/machine/fuji.conf:require conf/machine/include/ast2620.inc
meta-facebook/meta-grandcanyon/conf/machine/grandcanyon.conf:require conf/machine/include/ast2620.inc

I don't see cloudripper-image, fuji-image or grandcanyon-image targets.
I have an AST2600 eval board coming in and would like to know how to properly build one of these images.

I tried:
source openbmc-init-build-env meta-facebook/meta-fuji
but
bitbake fuji-image
.
.
WARNING: No recipes available for:
/localdata/dagilson/openbmc/yocto/warrior/poky/../../../common/recipes-tpm2/tpm2-tools/tpm2-tools_3.2.0.bbappend
ERROR: Nothing PROVIDES 'fuji-image'

What is the proper method to build one of the AST2600 images?  

Thanks in advance. 
...Dan Gilson

twlevin, 2020-10-05:
I am having the same issue when building for meta-portwell/meta-pwneptune except for me it is an ERROR:
ERROR: No recipes available for:
  .../openbmc/yocto/zeus/poky/../../../common/recipes-tpm2/tpm2-tools/tpm2-tools_3.2.0.bbappend
.../openbmc/yocto/zeus/poky/../../../meta-portwell/recipes-extended/rsyslog/rsyslog_8.18.0.bbappend

williamspatrick, 2020-10-20:
Those "no recipes available for" are warnings only and shouldn't cause any real issues.  Have you done something to increase the warning level?

I just now ran:
```
git clone ssh://git@github.com/facebook/openbmc
cd openbmc
./sync_yocto.sh
source openbmc-init-build-env fuji
bitbake fuji-image
```
And this was successful at starting a build.  If it doesn't work for you today, can you let me know more details about what distro you are running, etc.?

closed 2020-10-20 by williamspatrick