facebook/openbmc #159: Flashy will break on AST2600

lhl2617, 2021-04-29:
https://github.com/facebook/openbmc/commit/127f18d3a767aa2830286dbd32d6f1d3c5ef1fae points out that different SOC has different address space, and thus `AST_SRAM_VBS_BASE` needs to be defined differently. 

In `flashy`, `vboot-util` is "built-in"--the logic is replicated within the code, and can be seen here: https://github.com/facebook/openbmc/blob/helium/tools/flashy/lib/utils/vboot.go.
This replication, as recommended by @lsiudut et al., is due to early `vboot-util` bugs causing flash failures--I do not recall specifics, but it most likely is related to this commit: https://github.com/facebook/openbmc/commit/164ec0b60135143b0ed7a41be6aa4fee2dde71b6.
The major benefit of having `vboot-util` logic within `flashy` ties to `flashy`'s motivation itself: the ability of applying hotfixes within `flashy`'s code to flash devices with outdated utilities (incl. `vboot-util`) causing failing flashes.

The problem arises here: `AST_SRAM_VBS_BASE` is hardcoded in `flashy`, and it'd be hard to replicate the logic:
https://github.com/facebook/openbmc/blob/1db223dcfa849bc3d71b5f10548ab659f2dd1f3b/tools/flashy/lib/utils/vboot.go#L33

As of now, I think `vboot-util` should be stable enough, so the trivial fix is to rely on `vboot-util` within the system itself and rip the code off `flashy`, but the risk of such utilities causing large-scale flash failures still exist.

A better fix would be to somehow dynamically assign `AST_SRAM_VBS_BASE` in `flashy`. Opening this issue to discuss further on how we can do this.

lhl2617, 2021-04-29:
cc: @zhdaniel12 @kawmarco

lhl2617, 2021-04-29:
I researched further: perhaps `flashy` should also have its own `soc-util`?

https://github.com/facebook/openbmc/blob/helium/meta-aspeed/recipes-utils/soc-utils/files/soc-utils.sh

zhdaniel12, 2021-05-03:
thanks @lhl2617, recently we landed 1d35aa0af2e20b01a435634babd1ebb28ec73182, thus flashy shall be able to fix flashy.

lhl2617, 2021-05-04:
@zhdaniel12 Working on a fix :) Thanks!

lsiudut, 2021-05-04:
@lhl2617 I don't think w should rely on the new structures to detect the system as it won't work on the systems that are not having it implemented. This applies to golden images across the fleet.

lhl2617, 2021-05-04:
@lsiudut good point, I'm looking at it now. My plan was to fall back to the default one for AST2400/AST2500. 

I can however read off the device tree as in https://github.com/facebook/openbmc/commit/1d35aa0af2e20b01a435634babd1ebb28ec73182