facebook/openbmc #200: flash image

luoshuai777, 2022-08-11:
I want to flash my image, but I don't know which one to flash:

luoshuai777, 2022-08-11:
this is my Image directory: picky@ubuntu-local:~/FacebookBmc/openbmc/build/tmp/deploy/images/emeraldpools$ ls
emeraldpools-image-emeraldpools-20220811021842.rootfs.manifest
emeraldpools-image-emeraldpools-20220811021842.rootfs.tar.gz
emeraldpools-image-emeraldpools-20220811021842.testdata.json
emeraldpools-image-emeraldpools.manifest
emeraldpools-image-emeraldpools.tar.gz
emeraldpools-image-emeraldpools.testdata.json
fit-emeraldpools-20220811021842.itb
fit-emeraldpools.itb
flash-emeraldpools
flash-emeraldpools-20220811021842
modules--5.0.3-r1-emeraldpools-20220811021842.tgz
modules-emeraldpools.tgz
my-aspeed-bmc-facebook-emeraldpools--5.0.3-r1-emeraldpools-20220811021842.dtb
my-aspeed-bmc-facebook-emeraldpools.dtb
my-aspeed-bmc-facebook-emeraldpools-emeraldpools.dtb
u-boot.bin
u-boot-emeraldpools.bin
u-boot-emeraldpools-v2019.04-r0.bin
u-boot-fit-emeraldpools-20220811021842.itb
u-boot-fit-emeraldpools.itb
u-boot-recovery.bin
u-boot-recovery-emeraldpools.bin
u-boot-recovery-emeraldpools-v2019.04-r0.bin
u-boot-spl.bin
u-boot-spl-emeraldpools.bin
u-boot-spl-emeraldpools-v2019.04-r0.bin
zImage
zImage--5.0.3-r1-emeraldpools-20220811021842.bin
zImage-emeraldpools.bin

williamspatrick, 2022-08-29:
> flash-emeraldpools

closed 2022-08-29 by williamspatrick