facebook/openbmc #233: Cannot build yosemite project

NguyenTanNhutQuang, 2024-03-18:
Hi,

I have followed the build instructions to build the project yosemite
`
Note: In the instruction set below, references to for some of the steps is an example only and need to be replaced with the respective platform when setting up for a different platform.

Set up the build environment based on the Yocto Project's [Quick Start Guide](https://www.yoctoproject.org/docs/2.5/brief-yoctoprojectqs/brief-yoctoprojectqs.html).

Clone the OpenBMC repository and other open source repositories:

$ git clone -b helium https://github.com/facebook/openbmc.git
$ cd openbmc
$ ./sync_yocto.sh
Initialize a build directory for the platform to build. In the openbmc directory:
$ source openbmc-init-build-env wedge
Choose between wedge, wedge100, yosemite, or any of the other platforms listed in the meta-facebook directory. After this step, you will be dropped into a build directory, openbmc/build.

Start the build within the build directory: In general to build for the platform:
$ bitbake <platform>-image
The build process automatically fetches all necessary packages and builds the complete image. The final build results are in openbmc/build/tmp/deploy/images/<platform>. The root password will be 0penBmc, you may change this in the local configuration.
`
But it seems this instruction missing poky, meta-openembedded, and meta-security therefore it caused alot of errors during build.

I added those meta layers but the build still have had alot of errors most of them are missing file error. Could someone please point me to the correct intruction to build yosemite project?

Thank you
Quang

williamspatrick, 2024-03-27:
Yosemite is a very very old machine at this point.  I would not recommend attempting to build it.  What are you trying to achieve?

Can you give exact errors you are seeing and exactly what you've ran?

We do still build yosemite in CI internally, so I'm not sure what the issue might be.

PyMario, 2024-07-02:
> Yosemite 目前已经是一台非常老旧的机器了。我不建议尝试建造它。你想实现什么目标？
> 
> 您能否具体说明您所看到的错误以及您所运行的具体操作？
> 
> 我们确实仍在 CI 内部构建 yosemite，所以我不确定问题可能是什么。

I also encountered a similar problem and created issues. 
[#issue-2385736745](https://github.com/facebook/openbmc/issues/240#issue-2385736745)